xbst/KUSBA #2: Perhaps dumb question - will this work with a laptop running Ubuntu?

wiilover22, 2022-10-14:
This is probably a really stupid question - but since RPI's are so scarce right now, I'm running Klipper on an old laptop of mine sitting next to my 3D printer. 

Which is why I went looking for USB connection ADXL345's. 

If I already have Klipper installed on the Laptop, and my 3D printer flashed with the Klipper firmware - if I buy this USB ADXL345, is it just plug, config, and play? Does anything need to change? Thanks! Would love to buy one if it works with my laptop >.<

xbst, 2022-10-14:
It should work work fine, it's the same software. One thing to keep in mind is KUSBA is not compatible with USB C to C cables, so you need a USB A port on your laptop.

wiilover22, 2022-10-14:
So since I have everything installed already, it's just plug - setup the config file to read from the USB port - and run the resonance tests? Or do I have to recompile Klipper for my laptop or printer...or something?

Also...I have a bed slinger printer...does that mean I have to buy 2 of these so I can read the resonances from both the bed and printer head - if you happen to know...?

xbst, 2022-10-15:
You can just follow the instructions on this repo. I'm not sure where the wget command will download the cfg so you might have to manually move it. Other than that it should be the same.
You can buy just 1 and move the accelerometer, but if you want to measure both with 1 command or if you don't want to move it you will need 2.

wiilover22, 2022-10-15:
Thanks for the help!

closed 2022-10-15 by wiilover22