xbst/KUSBA #3: KUSBA with standard parts

Ritterbrater, 2022-10-30:
Hi,
nice project!
Instead of using a custom PCB I would like to take "standard" parts.
Is it possible to take a regular Raspberry Pi Pico Board with a ADXL345 module board? Do I only need to connect the pins GPIO0 - GPIO4 and then flash your firmware?

Another question: Are there any limitiations for the frequency measurement due to protocol change SPI <-> USB?

Thank you.

xbst, 2022-10-31:
It's possible. There are a few guides for that on the internet, I think Nero3D has a good video on that topic. 

The protocol between the accelerometer and the MCU is still SPI, performance should be the same.

Ritterbrater, 2022-10-31:
Ah, ok, I see. E.g.: https://www.youtube.com/watch?v=W_VHbT_tsZw
Thank you for the explanations.

closed 2022-10-31 by Ritterbrater