xbst/KUSBA #4: documentation of the protocol

tcurdt, 2022-11-07:
Fantastic project.
I was thinking of building something similar for another context (non-Klipper).

I am trying to understand how the communication works over USB.
What's the protocol? Are there some docs?
Is the latency a problem for the measurements?

Where is the code that actually runs on the MCU?
And what's the flash used for?

xbst, 2022-11-07:
The protocol is spi to the rp2040, then usb, so no different from the normal klipper setup. Latency shouldn't be any different compared to wiring the spi pins to a normal controller or a pi pico. The code is klipper, built at the time of flashing with the latest files available at the time.  flash is used to store the klipper code.

tcurdt, 2022-11-07:
> The protocol is spi to the rp2040

of course

> then usb

And that's the bit I am after.
But it sounds like this might be more of a question for the Klipper project.

To give more context:
Let's say I connect board and connect it to linux machine.
How could I talk to the board (without using Klipper)?

Seems I would need a custom firmware for the rp2040 and something on the host as well.

xbst, 2022-11-08:
Probably something can be programmed for that with arduino IDE.

closed 2022-11-12 by xbst