xbst/KUSBA #6: `Invalid adxl345 id (got f2 vs e5)` error running `ACCELEROMETER_QUERY` for the first time

hakimio, 2022-12-12:
I understand that `Invalid adxl345 id (got f2 vs e5)` error is documented but instead of ignoring it, it should be fixed. This doesn't happen with `FYSETC Portable Input Shaper` and should not happen with `KUSBA`.

xbst, 2022-12-13:
This happens with a Raspberry Pi Pico with ADXL345 too, not just KUSBA. This is a Klipper problem and there is nothing I can do to fix it.

hakimio, 2022-12-13:
- Why doesn't this happen with `FYSETC Portable Input Shaper` which also uses `RP2040`?
- If it's Klipper bug, where is the bug report and, if you know what's causing the issue, why don't you create a PR for Klipper?

hakimio, 2022-12-13:
I guess you were referring to this closed bug report: https://github.com/Klipper3d/klipper/issues/4712

xbst, 2022-12-13:
I have no idea why FYSETC PIS doesn't have this problem. Me and other people (simple discord searches can find some recent examples) had the invalid ID issue on the first query with Pi Picos too, not just KUSBA. This indicates a software issue.

One difference between the KUSBA vs PIS is, PIS uses different GPIO pins for the SPI bus to the ADXL, could be related. But the pins the KUSBA use are full HW SPI pins too, and they should work the same. Again, the only possible explanation I can think of for this is a Klipper bug, even if it is not reported on GitHub.

hakimio, 2022-12-13:
Can you create a bug report on Klipper repo and link it here?

Fragmon, 2023-01-04:
Hey, i try to create a macro that use the query for the test. But klipper dont stop the execution between the macros. Do you know a solution?

   ACCELEROMETER_QUERY
   G4 P10000
   TEST_RESONANCES AXIS=X

Fragmon, 2023-01-04:
I found a solution:

Use this:
[adxl345]
cs_pin: adxl:gpio1
#spi_bus: spi0a
spi_software_sclk_pin: adxl:gpio2
spi_software_mosi_pin: adxl:gpio3
spi_software_miso_pin: adxl:gpio0

xbst, 2023-01-12:
Ideally I prefer hardware SPI over software SPI, but I'll test it and if this fixes the issue without producing worse results I'll add it as an option to the .cfg.

closed 2023-02-02 by xbst

Witawat, 2023-02-24:
Invalid adxl345 id (got f2 vs e5).
This is generally indicative of connection problems
(e.g. faulty wiring) or a faulty adxl345 chip.

how to fix ?
or use klipper version ?

xbst, 2023-02-24:
It is covered in the instructions: https://github.com/xbst/KUSBA#note-the-first-reading-from-the-accelerometer-will-be-invalid-usually-f2-vs-e5-this-is-expected-and-your-kusba-will-still-work-fine-after-the-first-query-run-accelerometer_query-once-before-starting-measuring-resonances
First reading will be invalid, should work fine after that.