xbst/KUSBA #9: Unknown spi_bus 'spi0a'

mmithoefer, 2023-01-26:
I flashed my KUSBA and adjusted my serial address to : serial: /dev/serial/by-id/usb-Klipper_rp2040_E661347507974224-if00 which i determined by using ls /dev/serial/by-id/*.
When i try to start klipper i get  the following error:

Unknown spi_bus 'spi0a'

xbst, 2023-01-29:
You don't add the mcu to the printer.cfg, the mcu is in the adxlmcu.cfg. You just need to change the address in that file.

mmithoefer, 2023-01-29:
I´m aware of that. I did that only as a trouble-shooting measure. Whenever
 `[adxl345]
cs_pin: adxl:gpio1
spi_bus: spi0a`
is uncommented i get the described error

xbst, 2023-01-30:
That is weird. Can you update klipper? It sounds like a software issue since spi0a is a bus defined in the Klipper code for the RP2040.

mmithoefer, 2023-01-30:
I just did. Same error unfortunately

xbst, 2023-02-01:
Can you attach your printer.cfg and all .cfg files included with [include]?

mmithoefer, 2023-02-01:
Found the problem. Mistake on my part. I had a adxl section in my printer.cfg from previous experiments with other accelerometers.

closed 2023-02-01 by mmithoefer