xbst/KUSBA #11: 2.3 board loses flash if disconnected

cyborgcnc, 2023-02-11:
I have a 2.3 board, and it works as it should when flashed, however, if I disconnect the board, the flash file is no longer there....I have to go through the steps of mounting the file system again, and re-flashing, and then it works again...I thought this was supposed to be fixed in 2.3, but I am finding it is still there....wondering if I am doing something wrong?  Accelerometer works as it should once flashed, Klipper sees it etc.

xbst, 2023-02-11:
This is a known issue even with v2.3 but only affects about 1-2% of the PCBs according to data I get from vendors. Most of these are caught by the vendor before shipping to the customer but sometimes a faulty one is accidentally shipped. It sounds like you might have one of these PCBs unfortunately. Try contacting the seller.

cyborgcnc, 2023-02-11:
OK, I will do that...in the mean time, I guess I will keep re-flashing it for when I want to use it....royal  pain in the you know what....Thanks for your response!

closed 2023-02-11 by cyborgcnc

pellcorp, 2023-11-09:
I have this exact issue with the 2.4 board as well :-(