xbst/KUSBA #12: Trouble with make menuconfig

jperquin, 2023-02-17:
Have Fluiddpi / Moonraker Klipper installed on a Rpi 3b (BCM2835) and has been running fine with my Creality Ender6 that has been Klipper-ized with a custom flavour (https://3dprintbeginner.com/how-to-install-klipper-on-creality-ender-6/)

The KUSBA 2.3 is supposed to come pre-flashed with Klipper but does not show up under ls /dev/serial/by-id/*. The Ender6 does however (/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0). 

So figured I need to (re)flash the KUSBA, but following the instructions, I am running into an issue with make menuconfig.

I am supposed to be able to select these options:
```
Micro-controller Architecture: Raspberry Pi RP2040
Communication inferface: USB
```
But I am getting this of options for Micro-controller Architecture:

```
( ) Atmega AVR
( ) SAM3/SAM4 (Due and Duet)
( ) SAMD21/SAMD51
( ) LPC176x (Smoothieboard)
( ) STMicroelectronics STM32
( ) Beaglebone PRU
(X) Linux process
( ) Host simulator
```
So no "Raspberry Pi RP2040"..

Not sure how to proceed..

Your help is much appreciated.
Rgds, JP

xbst, 2023-02-17:
When did you last update Klipper? RP2040 is a newer chip than those so you might be running a really outdated version, from before RP2040 support. Try updating Klipper.

xbst, 2023-03-02:
Solved in another issue.

closed 2023-03-02 by xbst