xbst/KUSBA #13: KUSBA not recognized on USB

DK4FT, 2023-02-26:
received my KUSBA from lab4450.com a while ago and try to include it in my system now without success.

When pluggin in(with button pressed) the device is not recognized as a drive on my RPi.
As lab4450 mention on their page, that their devices are pre-flashed with the klipper firmware I tested also without button pressed, but the KUSBA is not regognized.

Here snippets from the syslog on the klipper machine..

1. KUSBA just pluggged in:
```
Feb 26 09:19:06 klipper kernel: [321786.579968] usb 1-1.3: new full-speed USB device number 7 using xhci_hcd
Feb 26 09:19:07 klipper kernel: [321786.680213] usb 1-1.3: device descriptor read/64, error -32
Feb 26 09:19:07 klipper kernel: [321786.900097] usb 1-1.3: device descriptor read/64, error -32
Feb 26 09:19:07 klipper kernel: [321787.020502] usb 1-1-port3: attempt power cycle
Feb 26 09:19:08 klipper kernel: [321787.679929] usb 1-1.3: new full-speed USB device number 8 using xhci_hcd
Feb 26 09:19:08 klipper kernel: [321787.680199] usb 1-1.3: Device not responding to setup address.
Feb 26 09:19:08 klipper kernel: [321787.900078] usb 1-1.3: Device not responding to setup address.
Feb 26 09:19:08 klipper kernel: [321788.119873] usb 1-1.3: device not accepting address 8, error -71
Feb 26 09:19:08 klipper kernel: [321788.219886] usb 1-1.3: new full-speed USB device number 9 using xhci_hcd
Feb 26 09:19:08 klipper kernel: [321788.220172] usb 1-1.3: Device not responding to setup address.
```

2. Button pressed while pluggin in: 
```
Feb 26 09:29:39 klipper kernel: [322419.318065] usb 1-1.3: new full-speed USB device number 10 using xhci_hcd
Feb 26 09:29:39 klipper kernel: [322419.418291] usb 1-1.3: device descriptor read/64, error -32
Feb 26 09:29:40 klipper kernel: [322419.638243] usb 1-1.3: device descriptor read/64, error -32
Feb 26 09:29:40 klipper kernel: [322419.858042] usb 1-1.3: new full-speed USB device number 11 using xhci_hcd
Feb 26 09:29:40 klipper kernel: [322419.958247] usb 1-1.3: device descriptor read/64, error -32
Feb 26 09:29:40 klipper kernel: [322420.178245] usb 1-1.3: device descriptor read/64, error -32
Feb 26 09:29:40 klipper kernel: [322420.298645] usb 1-1-port3: attempt power cycle
Feb 26 09:29:41 klipper kernel: [322420.968050] usb 1-1.3: new full-speed USB device number 12 using xhci_hcd
Feb 26 09:29:41 klipper kernel: [322420.968288] usb 1-1.3: Device not responding to setup address.
Feb 26 09:29:41 klipper kernel: [322421.188352] usb 1-1.3: Device not responding to setup address.
Feb 26 09:29:41 klipper kernel: [322421.408061] usb 1-1.3: device not accepting address 12, error -71
Feb 26 09:29:41 klipper kernel: [322421.508073] usb 1-1.3: new full-speed USB device number 13 using xhci_hcd
Feb 26 09:29:41 klipper kernel: [322421.508308] usb 1-1.3: Device not responding to setup address.
Feb 26 09:29:42 klipper kernel: [322421.728244] usb 1-1.3: Device not responding to setup address.
Feb 26 09:29:42 klipper kernel: [322421.948061] usb 1-1.3: device not accepting address 13, error -71
Feb 26 09:29:42 klipper kernel: [322421.948628] usb 1-1-port3: unable to enumerate USB device
Feb 26 09:29:53 klipper kernel: [322433.158187] usb 1-1.3: new full-speed USB device number 14 using xhci_hcd
Feb 26 09:29:53 klipper kernel: [322433.258356] usb 1-1.3: device descriptor read/64, error -32
Feb 26 09:29:53 klipper kernel: [322433.478410] usb 1-1.3: device descriptor read/64, error -32
Feb 26 09:29:54 klipper kernel: [322433.698219] usb 1-1.3: new full-speed USB device number 15 using xhci_hcd
Feb 26 09:29:54 klipper kernel: [322433.798392] usb 1-1.3: device descriptor read/64, error -32
Feb 26 09:29:54 klipper kernel: [322434.018417] usb 1-1.3: device descriptor read/64, error -32
Feb 26 09:29:54 klipper kernel: [322434.138656] usb 1-1-port3: attempt power cycle
Feb 26 09:29:55 klipper kernel: [322434.798216] usb 1-1.3: new full-speed USB device number 16 using xhci_hcd
Feb 26 09:29:55 klipper kernel: [322434.798449] usb 1-1.3: Device not responding to setup address.
Feb 26 09:29:55 klipper kernel: [322435.018475] usb 1-1.3: Device not responding to setup address.
Feb 26 09:29:55 klipper kernel: [322435.238283] usb 1-1.3: device not accepting address 16, error -71
Feb 26 09:29:55 klipper kernel: [322435.338247] usb 1-1.3: new full-speed USB device number 17 using xhci_hcd
Feb 26 09:29:55 klipper kernel: [322435.338476] usb 1-1.3: Device not responding to setup address.
Feb 26 09:29:55 klipper kernel: [322435.558444] usb 1-1.3: Device not responding to setup address.
Feb 26 09:29:56 klipper kernel: [322435.778234] usb 1-1.3: device not accepting address 17, error -71
Feb 26 09:29:56 klipper kernel: [322435.778773] usb 1-1-port3: unable to enumerate USB device
```

in both cases the device is not shown (as expected after these log messages.
```
pi@klipper:~ $ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
```

Is there a chance to reacitvate the KUSBA, or is it a DOA case where I need to open a ticket with LAB4450?

xbst, 2023-02-26:
Did you try using a different USB A to C cable? Some cables only have the power wires and not the data wires. I'm not a Linux expert so I'm not a 100% sure if these errors would show up if a device is connected with a power-only cable. Also, C to C cables aren't supported. If a C to C cable is used, the data pins are connected but no power is provided to the KUSBA (v2.3).

Try using a different A to C cable with the data wires, while pressing the button. If it shows up as a drive, you can flash it again and use it with that cable. Since LAB4450 tests each KUSBA before shipping, I would expect it to work fine but if it doesn't work after testing a few USB cables it probably makes sense to contact them.

DK4FT, 2023-02-27:
Thanks for the recommendations - I close the issue here, as it seems to be HW related...

I tested different USB A to USB C cables, which I already used to sync phones or connected other devices.
As the KUSBA is nor shown on a Linux nor on a windows machine as a drive, i assume there is something wrong with the device and contact LAB4450.

I will followup with LAB4450...

closed 2023-02-27 by DK4FT

DK4FT, 2023-02-28:
Lab4450 were very nice and assuming a bricked KUSBA device as well - they are sending a new one...

jperquin, 2023-03-01:
I have a similar issue, but AM able to copy compiled FW after connecting with reset button pressed:

```
sudo mount /dev/sda1 /mnt
sudo cp out/klipper.uf2 /mnt
sudo umount /mnt
```
Rebooting my main mcu or reconnecting the KUSBA does not help..

xbst, 2023-03-01:
> I have a similar issue, but AM able to copy compiled FW after connecting with reset button pressed:
> 
> ```
> sudo mount /dev/sda1 /mnt
> sudo cp out/klipper.uf2 /mnt
> sudo umount /mnt
> ```
> 
> Rebooting my main mcu or reconnecting the KUSBA does not help..

You already have an issue open. Please don't post on other issues as it makes it difficult to keep track of it and offer help. Can you clarify the current problem in the issue you opened? Sounds like you now can flash the firmware to the KUSBA but not seeing the serial address?

jperquin, 2023-03-02:
Yeah sorry about that. I updated my klipper and was able to compile and upload the FW for the Rpi RP2040 to volume sda1, which makes me believe that the physical connection is OK.
So yes, you are right that I cannot see the serial address (eg. /dev/serial/by-id/usb-Klipper_rp2040_E6622C42E359A225-if00).

Your help is much appreciated.

jperquin, 2023-03-02:
One more reconnecting of the KUSBA appears to have solved the issue. :-)

```
pi@fluiddpi:~ $ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0  /dev/serial/by-id/usb-Klipper_rp2040_E6622C42E359A225-if00
```

grahamr, 2023-08-10:
> Lab4450 were very nice and assuming a bricked KUSBA device as well - they are sending a new one...

Did the new one work? I found this older post trying to get mine working as well - tried 2 cords, a pi3b and a BTT CB1, and a windows box and macbook pro with no luck

DK4FT, 2023-08-10:
Yea, the new one worked like a charm.
The old kusba was simply doa by some unknown reason