xbst/KUSBA #17: Confused about orientation

bobg2023, 2023-03-23:
Hi,

I have a bed slinger ( Ender 3 ) I am about to setup my Kusba on.

I am going to use the mk6 nozzle mount for the hotend( X) axis and another mount for the bed (Y)

I have heard it said that you don’t have to worry about the orientation of how the Kusba is mounted because klipper just combines the values anyway, but from looking at the graphs produced  they are graphed separately and the m6 nozzle mount on this site says 

‘ Add "axes_map: y, x, z" to [adxl345] section config for proper orientation with the USB-C facing forward.’

So I’m confused as to whether orientation matters when running SHAPER_CALIBRATE AXIS=X 

Thank you

xbst, 2023-03-23:
It doesn't matter. That mount was contributed by another user, so that description was written by them, and I didn't edit it just in case someone wants to enter the "correct" axes map for that mount.
Klipper ignores the axes map for its calculations, like you said. The axes map only is used for visually marking each axis for the graphs generated by Klipper.

bobg2023, 2023-03-23:
Thank you for explaining.

closed 2023-03-23 by bobg2023