xbst/KUSBA #25: Kuba

pimp1310, 2023-08-19:


Alessiopell90, 2023-09-06:
Ciao, ho esattamente lo stesso problema tuo. Dopo vari tentativi oggi sono riuscito a fare l'input shaping però dal momento che spengo e riaccendo la stampante non mi trova più il seriale. Tu hai risolto?

Johnmkar, 2023-09-11:
I have the same problem. The KUSBA does not show up when I plug it in holding the boot button in. The ls /dev/ command results are identical with and without it plugged in. Unfortunately the KUBSA does not have any LED showing the unit is actually powered on so I can't even tell if it is working. I'm running Klipper on a BTT Pi v1.2 with a HDMI5 screen and I'm not sure if either one could be causing the issue.

Alessiopell90, 2023-09-11:
hi, I solved the problem. in the cfg printer I had the configuration of
another accelerometer and therefore they conflicted, after which I also
made a mistake in pressing the button when I wanted to make it work.
therefore, I insert the USB cable without pressing the button, I remove the
# from the cfg printer, I do shaper_calibrate, the first time it gives an
error and the second time it starts working.

xbst, 2023-09-11:
> I haven't gotten to the point of configuring the accelerometer in printer.cfg. There is nothing configured for another one. I can't located it when ssh into the Pi and do the ls/dev/ command so I can flash the firmware. That's why I'm wondering if the board is dead or there is a conflict with my HDMI5 screen, since the only USB listed is TTYUSB0, and it's shown regardless of whether or not the KUSBA is plugged in or not. I would expect TTYUSB1 to show up.

I can't comment on the potential display conflict as I don't have a setup like that, but one of the first steps of troubleshooting issues like this is trying another USB cable. A lot of USB C cables don't have the data pins so it can't communicate with the Pi. I'd also try connecting it to a different computer to verify if the issue is caused by the setup with the display or cable or bad KUSBA. On a Windows PC, unflashed KUSBA will look like a USB drive, flashed one will open a serial (COM) port, which should show up in device manager.

Alessiopell90, 2023-09-11:
I too have tried different cables, however I am noticing that every now and
then I connect the USB cable and there is no signal, I plug it in and out
2/3 times and it goes. it is a strange thing. Luckily it works but it's not
accurate on the first try

pellcorp, 2023-11-09:
So I can only get this to work if I copy the firmware over every time I plug it in.  So I need to hold the button down while plugging it in, then do these commands:

```
pi@klipper:~/klipper $ sudo mount /dev/sda1 /mnt
pi@klipper:~/klipper $ sudo cp out/klipper.uf2 /mnt
pi@klipper:~/klipper $ sudo umount /mnt
```

And the `ls /dev/serial/by-id/` shows the device, but if then unplug the device and then plug it in again, the device does not appear.