xbst/KUSBA #26: Invalid adxl345 id (got ff vs e5)

DarkstarDev, 2023-10-23:
**The problem**: Running `ACCELEROMETER_QUERY` yields the following every time:
```
// Invalid adxl345 id (got ff vs e5).
// This is generally indicative of connection problems
// (e.g. faulty wiring) or a faulty adxl345 chip.
!! Invalid adxl345 id (got ff vs e5).
```

**What I've tried**:
* Verified communication with the printer
  * KUSBA device is listed in **/dev/serial/by-id/**
  * Disconnecting the KUSBA module crashes Klipper (`!! Lost communication with MCU 'adxl'`)
  * Restarting Klipper on the printer with KUSBA enabled in **printer.cfg** continually crashes Klipper (`!! mcu 'adxl': Unable to connect`)
  * Commenting out the **include** line in **printer.cfg** and restarting Klipper on the printer restores normal functionality of the printer
  * Ensured I was following the [KUSBA v2 Instructions for Rampon](/xbst/KUSBA/blob/main/Docs/v2-Rampon-Firmware.md) (KUSBA came with Rampon installed)
* Successfully flashed Klipper onto KUSBA
* Followed [Kusba v2 Klipper Instructions](/xbst/KUSBA/blob/main/Docs/v2-Firmware.md) and edited **adxlmcu.cfg** according to the [Klipper config version](/xbst/KUSBA/blob/main/Firmware/v2/adxlmcu.cfg) (see latest version below)
* Updated **adxlmcu.cfg** to use software SPI, but received the error `adxl345: spi pins must be on same mcu`
* Tried multiple USB-C cables with the same results

**Additional information**
I tried running `ACCELEROMETER_QUERY` with KUSBA unplugged and disabled in configs. Received the same error (`Invalid adxl345 id (got ff vs e5).`). This seems to suggest that `ff` is the default state/response and I'm unable to get a successful response from the accelerometer. The fact that the USB device appears, can be flashed, and responds enough that Klipper doesn't crash when booting and communicating with the KUSBA seems to indicate a problem with the accelerometer on the KUSBA, but that's just my guess.

---
**~/klipper_config/printer.cfg**:
```
# This file contains common pin mappings for ELEGOO
# boards. To use this config, the firmware should be compiled for the
# stm32f407. When running "make menuconfig", select the 48KiB
# bootloader, and enable "Serial for communication" and select the "on USART1 PA10/PA9"
# See docs/Config_Reference.md for a description of parameters.

[include plr.cfg]
[include adxlmcu.cfg]
... more printer config here ...
```

**~/klipper_config/adxlmcu.cfg**:
```
[mcu adxl]
serial: /dev/serial/by-id/usb-Klipper_rp2040_E6624C90272F2721-if00
# Edit the above line with the correct serial address. Use "ls /dev/serial/by-id/*" to find the MCU serial address.

[adxl345]
cs_pin: adxl:gpio1

# Hardware SPI Option (Better results, "Invalid ID" Klipper bug with first query)
spi_bus: spi0a

# Software SPI Option (No "Invalid ID" Klipper bug with first query)
# spi_software_sclk_pin: adxl:gpio2
# spi_software_mosi_pin: adxl:gpio3
# spi_software_miso_pin: adxl:gpio0

[resonance_tester]
accel_chip: adxl345
probe_points:
   117,117,20
# Edit the above line with the correct probe points location. I recommend the center of your bed for X & Y, 20 for Z.
# For 350 mm printers: 175,175,20
# For 300 mm printers: 150,150,20
# For 250 mm printers: 125,125,20
# For 120 mm printers: 60,60,20

# More info: https://www.klipper3d.org/Config_Reference.html#adxl345
```

xbst, 2023-10-23:
Did you try Rampon before flashing Klipper? Rampon is the prefferred firmware because it is less buggy.

Can you flash Rampon again using the instructions here?:
https://github.com/xbst/KUSBA/blob/main/Docs/v2-Rampon-Firmware.md#1-flash-rampon-anchor-to-the-mcu
And use this adxlmcu.cfg:
https://github.com/xbst/KUSBA/blob/main/Firmware/v2-Rampon/adxlmcu.cfg

If this doesn't fix your issue, can you send your entire printer.cfg and other .cfgs included with [include]?

DarkstarDev, 2023-10-23:
While I did try initially following the instructions for Rampon, I did not try re-flashing Rampon, then configuring again. I can try that and see if anything changes.

DarkstarDev, 2023-10-23:
Unfortunately, the outcome appears to be the same after flashing and configuring. I verified the path to the USB device, and the printer boots fine with the device configured, but I still get `ff` as the response. As requested, here is my **printer.cfg** and the files it includes.

**printer.cfg**
```
# This file contains common pin mappings for ELEGOO
# boards. To use this config, the firmware should be compiled for the
# stm32f407. When running "make menuconfig", select the 48KiB
# bootloader, and enable "Serial for communication" and select the "on USART1 PA10/PA9"
# See docs/Config_Reference.md for a description of parameters.

[include plr.cfg]
[include adxlmcu.cfg]

[mcu]
# The hardware use USART1 PA10/PA9 connect to RK3328
serial: /dev/ttyS0
restart_method: command

[stepper_x]
step_pin:PC14
dir_pin:PC13
enable_pin:!PC15
microsteps: 16
rotation_distance: 40
full_steps_per_rotation:200  #set to 400 for 0.9 degree stepper
endstop_pin:PC0
position_min: -8  
position_endstop:-8
position_max: 235
homing_speed:50
homing_retract_dist:5
homing_positive_dir:false
step_pulse_duration:0.000002

[stepper_y]
step_pin:PB4
dir_pin:PB3
enable_pin:!PB5
microsteps:16
rotation_distance: 40
full_steps_per_rotation:200  #set to 400 for 0.9 degree stepper
endstop_pin:PB8
position_min: 0
position_endstop:0 
position_max:235
homing_speed:50
homing_retract_dist:5
homing_positive_dir:false
step_pulse_duration:0.000002

[stepper_z]
step_pin:PC10
dir_pin:!PA15
enable_pin: !PC11
microsteps: 16
rotation_distance: 8
full_steps_per_rotation: 200
endstop_pin:probe:z_virtual_endstop ## PB12 for Z-max; endstop have'!' is NO

##  Z-position of nozzle (in mm) to z-endstop trigger point relative to print surface (Z0)
##  (+) value = endstop above Z0, (-) value = endstop below
##	Increasing position_endstop brings nozzle closer to the bed
##  After you run Z_ENDSTOP_CALIBRATE, position_endstop will be stored at the very end of your config
#position_endstop:-5
position_max: 268
position_min: -5 
homing_speed: 8
second_homing_speed: 3

[extruder]
[... 503 lines cut ...]
    SAVE_VARIABLE VARIABLE=power_resume_z VALUE={z_pos}
[save_variables]
filename =/home/mks/klipper_config/saved_variables.cfg
```

**adxlmcu.cfg**
```
[mcu adxl]
serial: /dev/serial/by-id/usb-Anchor_Rampon-if00

[adxl345]
cs_pin: adxl:CS

[resonance_tester]
accel_chip: adxl345
probe_points:
   117,117,20
# Edit the above line with the correct probe points location. I recommend the center of your bed for X & Y, 20 for Z.
# For 350 mm printers: 175,175,20
# For 300 mm printers: 150,150,20
# For 250 mm printers: 125,125,20
# For 120 mm printers: 60,60,20

# More info: https://www.klipper3d.org/Config_Reference.html#adxl345
```

DarkstarDev, 2023-10-23:
I'm getting readings now. Thank you so much for your quick debugging on this. As a newbie to 3D printing, I didn't think to even check for conflicting configs since the Neptune 4 doesn't have a built-in accelerometer. I don't know what the best way to avoid this would be for future buyers, but a simple note in your documentation, or a troubleshooting guide might help.

Kudos for creating something that makes it so quick and easy to test, reflash, and reconfigure, and thank you again for your help.

closed 2023-10-23 by DarkstarDev

Spedcat, 2024-11-13:
> Did you try Rampon before flashing Klipper? Rampon is the preferred firmware because it is less buggy.
> 
> Can you flash Rampon again using the instructions here?: https://github.com/xbst/KUSBA/blob/main/Docs/v2-Rampon-Firmware.md#1-flash-rampon-anchor-to-the-mcu And use this adxlmcu.cfg: https://github.com/xbst/KUSBA/blob/main/Firmware/v2-Rampon/adxlmcu.cfg
> 
> If this doesn't fix your issue, can you send your entire printer.cfg and other .cfgs included with [include]?

Hi im having some troubles with installing Rampon like you said becuase it seems like its refering to another accelerometer than I have (btt adxl345) and it references a file named "sda1" which is a file i dont have on my pi am I doing somthing wrong here?