xbst/KUSBA #27: Kusba 2.4 instalation process

rastosk, 2023-10-28:
Hi i have received my kusba 2.4 from trianglelabs and i have to say what a beautiful work of art is this! When I was commissioning the accelerometer and following the guide:

  6.  Build the firmware

make

7.    Find the storage location of the KUSBA. This will usually be sda1. Use this command one time with the KUSBA unplugged and one time with KUSBA plugged in (while holding down the button on the KUSBA) to verify.

ls /dev/

   8. Flash the firmware.

sudo mount /dev/sda1 /mnt
sudo cp out/klipper.uf2 /mnt
sudo umount /mnt

Step 7 didn't make any sense to me so what i have done just skipped that step. Does this mean that my adxl will not work properly?
Thank you for your help!

Will-wastelander, 2023-10-28:
>    1. Build the firmware
>    1. Find the storage location of the KUSBA. This will usually be sda1.
>    1. Flash the firmware.

From the.looks, it's talking about the ls /dev/ command, so you can verify
the device ID is in fact sda1.

rastosk, 2023-10-28:
This what I have at L's /dev

serial: /dev/serial/by-id/usb-Klipper_rp2

Will-wastelander, 2023-10-28:
That would mean the device has already been flashed, and you will need to
hold the button while plugging in the USB cable to get to the bootloader.

rastosk, 2023-10-28:
Ok so just plug in hold the button and what do I look for?

Will-wastelander, 2023-10-28:
Exactly what you posted.

rastosk, 2023-10-28:
So when I hold button down it will boot differently? When I type Ls/dev what should come out?

xbst, 2023-10-31:
Just follow these instructions:

https://github.com/xbst/KUSBA/blob/main/Docs/v2-Rampon-Firmware.md#1-flash-rampon-anchor-to-the-mcu

You can flash the firmware on a Windows PC, it's easier.

rastosk, 2023-10-31:
I remember doing that with adxl345 board. But looks like this kusba come flashed ! Do they come flashed from factory? If so that must be very old firmware on it surprised that this actually works.

xbst, 2023-11-07:
There's no one "factory" for KUSBAs. I open sourced the files so there are many vendors, and each vendor orders from their preferred supplier. Some even make them themselves. Either way, they don't come with the firmware, it needs to be flashed. I flash all of mine before selling them. I know some other vendors do that too, but not all.
Rampon, unlike Klipper, doesn't need to be updated often. So if you bought from me or another vendor that pre-flashes rampon, it should be good to go. No update needed until there's a new rampon version.

pellcorp, 2023-11-09:
I actually had a problem getting mine to work, I have the 2.4 version 

So I can only get this to work if I copy the firmware over every time I plug it in. So I need to hold the button down while plugging it in, then do these commands:

pi@klipper:~/klipper $ sudo mount /dev/sda1 /mnt
pi@klipper:~/klipper $ sudo cp out/klipper.uf2 /mnt
pi@klipper:~/klipper $ sudo umount /mnt

And the ls /dev/serial/by-id/ shows the device, but if then unplug the device and then plug it in again, the device does not appear.

Am I doing it wrong, or is the device defective.

This is the same issue for the Rampon as well

rastosk, 2023-11-09:
Hi pellcorp just follow this video and you will be golden. Supper easy
[https://www.youtube.com/watch?v=W_VHbT_tsZw](url)

pellcorp, 2023-11-09:
So it actually seems like have this exact issue https://github.com/xbst/KUSBA/issues/11

@rastosk I followed these steps to the letter, the issue is just like for issue #11, after I disconnect the device from the computer the flashed firmware gets reset, so it seems like I have a faulty board, I am chasing up a replacement from my vendor, fingers crossed they do the right thing.

rastosk, 2023-11-09:
That sucks man I gotine from aliexpress trianglelabs

pellcorp, 2023-11-09:
Seems like it just a random issue, and Unique Prints Shop  have already created a order to replace mine (they will now test that the one they send me saves the flash image correctly) and in the mean time I can use I just have to the mount cp, umount steps first.

sreknob, 2023-11-28:
I just received one from Triangle Labs that has the exact same issue, will work after flashing but won't show up as a serial device after a power cycle. I presume there is probably an issue with pulling the chip into run mode.

rastosk, 2023-11-28:
> I just received one from Triangle Labs that has the exact same issue, will work after flashing but won't show up as a serial device after a power cycle. I presume there is probably an issue with pulling the chip into run mode.

Interesting mine work fine after reboot. Try to reflash and follow steps thatbinposted above

sreknob, 2023-11-30:
> Interesting mine work fine after reboot. Try to reflash and follow steps thatbinposted above

I have two from Triangle Labs, only one exhibits this behaviour. I have re-flashed several times, both from windows and linux with the same results. They came factory flashed with Rampon 0.2.0 and one didn't work immediately. Silkscreen says version 2.4. It was only after doing the reflash several times that I discovered this behaviour.

xbst, 2023-11-30:
I test all of my KUSBAs as this is a problem affecting a decent chunk of KUSBAs, depends on the batch but I see somewhere between 2-8% failure rate (usually closer to 2%). Here's a picture of all the KUSBAs I discarded and didn't sell in the past 6 months or so because of this issue:
[image]
Obviously discarding this many units is a big business expense, so they don't bother. And because of this and because I don't trust them to make it right when customers get a bad unit, I don't list them on the vendor list in this repo.

sreknob, 2023-12-01:
Fair enough! Next time I will order from you!

While trying to figure out what was going on, I had watched your video about the issues with the earlier revisions and the process to get to 2.3/2.4 so I guess I was just surprised that the flash erase issue was still a problem.

Thanks for all you do Isik!

xbst, 2023-12-02:
Yeah I thought the problem was fixed with v2.3 as all the units I ordered for the prototype run were working fine. I wasn't selling these back then so I didn't have a huge number of units to test.

Months later a few reports started coming from other vendors of 1-2% of units failing like that, and that is when I knew the issue wasn't fully fixed. I was still relying on other vendors reporting failed units to me at the time. When I started selling these about 6 months ago, I realized the fail percantage is actually 2-8%, depending on the batch. This high variability between batches is weird but I don't know what's going on exactly, just some theories. All I know for sure is, this only seems to happen on RP2040 based PCBs I design, which is part of the reason why I'm now using STM32s on newer projects.

Another observation is Rampon sometimes works fine on units that fail with Klipper. It's also less buggy on RP2040 (no failed first query bug), and doesn't require updates as often, and easier to use. So that's the firmware I recommend and the firmware I use to pre-flash/test my units.

rastosk, 2023-12-02:
The error that you get every time you do a query is same as Nero has in his video loose wire. The moment he puts wire in place and do query all is fixed . I'm wondering if the kusba issue could be related to that somehow. This is way beyond my knowledge just guessing here.

rastosk, 2023-12-02:
Oh cool how do I flash rampton? I don't mind the bug not a big deal but its a pain to reflash the klipper file after each upgrade

xbst, 2023-12-02:
https://github.com/xbst/KUSBA/blob/main/Docs/v2-Rampon-Firmware.md

austwhite, 2024-02-29:
I just received a KUSBA from local reseller DREMC .   I have run into the issue of the firmware not sticking and erasing on a power cycle, like it is loaded into RAM but never makes it's way to flash or something.
Anyway, I wanted to ask here before I go back to DREMC to see if there was anyway to get the firmware to stick on one that has issues?     DREMC are usually fantastic and easy to talk to, I just want to make sure I exhaust all options before I go back to them

xbst, 2024-02-29:
> I just received a KUSBA from local reseller DREMC . I have run into the issue of the firmware not sticking and erasing on a power cycle, like it is loaded into RAM but never makes it's way to flash [...]

Sometimes Rampon works fine on units that fail like that with Klipper, so if you're flashing Klipper, Rampon might help.

Other than this, not much can be done unfortunately. If it's bothering you, you need to contact DREMC for a replacement.

austwhite, 2024-02-29:
@xbst    More of a slight niggle than a bother.   Just means I have to flash it from the Linux CLI as if I dump the file from windows, of course it is gone when I disconnect it from the Windows machine.    I did try Rampon first to be honest.  No luck with my one :).
Thanks for the advice.   I appreciate it.