xbst/KUSBA #28: KUSBA accelerator board orientation

tkocou, 2023-11-22:
I have a Flsun V400 which I would like to be able to use the accelerator board. I was wondering about what the proper orientation of the board would be. I can CAD a mounting board myself, but need to know how to orient the KUSBA board with respect to X,Y,Z axis. Would the 2 mounting holes be in alignment in the Z axis, or is there a different orientation?

xbst, 2023-11-22:
The ADXL345 axes are printed on the PCB, on the side with the components, above KUSBA.

That being said, for Klipper input shaping applications, you don't need to match the accelerometer axes to printer axes. Klipper uses the sum of the data from all 3 of the accelerometer axes when it's doing its input shaper calculations.

tkocou, 2023-11-22:
I looked. You are correct. The axis are printed. However, the type is very small. It is hard to see which direction each axis has. Thank you for the information.

closed 2023-11-22 by tkocou