xbst/KUSBA #29: `Invalid adxl345 id (got 0 vs e5)`

Samywamy10, 2023-12-14:
Hi,
Recently got a Kusba from [Unique Prints in Australia](https://uniqueprints.shop/shop/electronics-electrical/pcb/kusba-usb-adxl345-accelerometer-for-klipper/) and I'm getting this `Invalid adxl345 id (got 0 vs e5)` error after using the Firmware v2 with Rampon.

I also tried non-Rampon and got the f2 error as described in other issues.

I've tried a whole bunch of different USB-C cables. In fact the one I'm using now I used to connect to the printer itself so its known working.

I can see the device is coming up in USB
```
sam@raspberrypi:~/adxl345spi $ lsusb
Bus 001 Device 005: ID 0451:8142 Texas Instruments, Inc. TUSB8041 4-Port Hub
Bus 001 Device 008: ID 1d50:614e OpenMoko, Inc. Rampon
Bus 001 Device 006: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 002: ID 0451:8142 Texas Instruments, Inc. TUSB8041 4-Port Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
sam@raspberrypi:~/adxl345spi $ ls /dev/serial/by-id/*
```

and as a device
```
sam@raspberrypi:~/adxl345spi $ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0  /dev/serial/by-id/usb-Anchor_Rampon-if00
```

I tried using this [accelerometer reader ](https://github.com/nagimov/adxl345spi)and got all 0s as well (but i had the same output when I unplugged the device so not sure how accurate this is):

```
sam@raspberrypi:~/adxl345spi $ sudo ./adxl345spi
time = 0.000, x = 0.000, y = 0.000, z = 0.000
time = 0.200, x = 0.000, y = 0.000, z = 0.000
time = 0.400, x = 0.000, y = 0.000, z = 0.000
time = 0.601, x = 0.000, y = 0.000, z = 0.000
time = 0.801, x = 0.000, y = 0.000, z = 0.000
time = 1.001, x = 0.000, y = 0.000, z = 0.000
time = 1.201, x = 0.000, y = 0.000, z = 0.000
time = 1.401, x = 0.000, y = 0.000, z = 0.000
time = 1.601, x = 0.000, y = 0.000, z = 0.000
time = 1.802, x = 0.000, y = 0.000, z = 0.000
time = 2.002, x = 0.000, y = 0.000, z = 0.000
time = 2.202, x = 0.000, y = 0.000, z = 0.000
time = 2.402, x = 0.000, y = 0.000, z = 0.000
time = 2.602, x = 0.000, y = 0.000, z = 0.000
time = 2.803, x = 0.000, y = 0.000, z = 0.000
time = 3.003, x = 0.000, y = 0.000, z = 0.000
time = 3.203, x = 0.000, y = 0.000, z = 0.000
time = 3.403, x = 0.000, y = 0.000, z = 0.000
time = 3.603, x = 0.000, y = 0.000, z = 0.000
time = 3.803, x = 0.000, y = 0.000, z = 0.000
time = 4.004, x = 0.000, y = 0.000, z = 0.000
time = 4.204, x = 0.000, y = 0.000, z = 0.000
time = 4.404, x = 0.000, y = 0.000, z = 0.000
time = 4.604, x = 0.000, y = 0.000, z = 0.000
time = 4.804, x = 0.000, y = 0.000, z = 0.000
25 samples read in 5.04 seconds with sampling rate 5.0 Hz
Done
```

**adxlmcu.cfg**
```
[mcu adxl]
serial: /dev/serial/by-id/usb-Anchor_Rampon-if00

[adxl345]
cs_pin: adxl:CS

[resonance_tester]
accel_chip: adxl345
probe_points:
  110,110,20
# Edit the above line with the correct probe points location. I recommend the center of your bed for X & Y, 20 for Z.
[... 1026 lines cut ...]
variable_probe_dock_enable: False           # Set to True to enable the usage of a dockable probe.      # ---------------------------------------------
variable_attach_macro: 'Attach_Probe'       # The macro that is used to attach the probe.               # Klicky Probe:   'Attach_Probe' | 'Dock_Probe'
variable_detach_macro: 'Dock_Probe'         # The macro that is used to store the probe.                # Euclid Probe:   'Deploy_Probe' | 'Stow_Probe'
                                                                                                        # Legacy Gcode:   'M401'         | 'M402'

# The following variables are for adjusting adaptive purge settings for KAMP.
variable_purge_height: 0.8                  # Z position of nozzle during purge, default is 0.8.
variable_tip_distance: 0                    # Distance between tip of filament and nozzle before purge. Should be similar to PRINT_END final retract amount.
variable_purge_margin: 10                   # Distance the purge will be in front of the print area, default is 10.
variable_purge_amount: 30                   # Amount of filament to be purged prior to printing.
variable_flow_rate: 12                      # Flow rate of purge in mm3/s. Default is 12.

# The following variables are for adjusting the Smart Park feature for KAMP, which will park the printhead near the print area at a specified height.
variable_smart_park_height: 10              # Z position for Smart Park, default is 10.

gcode: # Gcode section left intentionally blank. Do not disturb.

    {action_respond_info(" Running the KAMP_Settings macro does nothing, it is only used for storing KAMP settings. ")}

```

</p>
</details> 

Let me know what further info I can provide to help in debugging this!

xbst, 2023-12-31:
Hi,

First of all, sorry for taking this long to reply. I didn't get an email from GitHub for some reason (checked) so I just saw this issue when checking the repo.

0 is the response you'll get when there's no response. I'm not familiar with the non-Klipper tool you used, but it looks like it's for connecting ADXL345s to Pi's GPIO. Maybe I'm missing something but I don't think it'll work with the KUSBA. Your ADXL345 isn't connected to your Pi's GPIO so no signal, 0.

But you shouldn't get 0 with Klipper. Most `Invalid adxl345 id (got 0 vs e5)` issues are because of conflicting ADXL345 definitions in the `printer.cfg`. Your `printer.cfg` looks good but maybe you're including a wrong file with `[include]`? I noticed the `[include]` for macros is `[include macro.cfg]` but your macro file above is named `macros.cfg`. Maybe you have a `macro.cfg` somewhere and it has conflicting configs in it? Or maybe one of your KAMP configs has conflicting ADXL345 configs in it for some reason? It's a stretch but this is all I can think of as far as config issues go.

If you have a spare Pi or spare SD card, can you try a fresh install to be sure? Maybe it's a weird Klipper bug.

Also, you mentioned you tried Klipper firmware (on the KUSBA) too but got f2. Did you try doing a second `ACCELEROMETER_QUERY`? I'm guessing that'll also be `Invalid adxl345 id (got 0 vs e5)` like Rampon, but if not maybe it's a Rampon bug.

If nothing works, I guess the only other possibility is a hardware issue. Your MCU, flash and all the supporting chips for the MCU seem to be working since you can flash the firmware, so if there's a hardware issue, it has to be ADXL345 related. You might have a bad ADXL345 chip, or more likely have bad solder joints on the ADXL345.

Samywamy10, 2024-01-08:
Hi, thanks for your response

Apologies, my file is called `macro.cfg`. So I don't think its that.

> If you have a spare Pi or spare SD card, can you try a fresh install to be sure? Maybe it's a weird Klipper bug.

Unfortunately don't have a spare pi. I could try reinstalling Klipper I guess

> Did you try doing a second ACCELEROMETER_QUERY? I'm guessing that'll also be Invalid adxl345 id (got 0 vs e5) like Rampon, but if not maybe it's a Rampon bug.

Yeah I ran it multiple times with the same output. That's actually why I switched to Rampon.

Let me follow up with the seller to see if can try a different KUSBA.