xbst/KUSBA #30: ADXL error after updating klipper

faxalex, 2024-01-21:
Hi, I'm Alessio..

premise that before the klipper update everything worked properly

MCU Protocol error This is frequently caused by running an older version of the firmware on the MCU(s). Fix by recompiling and flashing the firmware. Your Klipper version is: v0.12.0-85-gd785b396-dirty MCU(s) which should be updated: mcu: Current version v0.12.0-85-gd785b396 adxl: Current version v0.12.0-58-g25bc649c Up-to-date MCU(s): <none> Once the underlying issue is corrected, use the "RESTART" command to reload the config and restart the host software. mcu 'adxl': Command format mismatch: query_adxl345 oid=%c rest_ticks=%u vs query_adxl345 oid=%c clock=%u rest_ticks=%u

I tried updating the firmware with the latest version but nothing changed.
Could an update be necessary on your part or is it a klipper bug?

Thanks in advange

xbst, 2024-01-21:
Apparently the latest Klipper update broke it. I'm not the dev of that firmware but I know they are already looking into it. You will need to reflash it with updated firmware when it gets released.

If you want to be able to use it in the meantime, Klipper firmware should still be working: https://github.com/xbst/KUSBA/blob/main/Docs/v2-Firmware.md

xbst, 2024-01-22:
New Rampon version supporting new Klipper:
https://github.com/rogerlz/rampon_anchor/blob/main/release/rampon_anchor_kusba.uf2

closed 2024-01-22 by xbst