xbst/KUSBA #31: Whenever I try to copy over the klipper.uf2 the KUSBA disconnects.

dvystrcil, 2024-02-03:
I have tried on my printer and my windows using WSL. 

- I plug it in while holding the button, I can see the device.
- I can mount the folder as per the instructions.
- As soon as I copy the file over it device disappears. 
- When I reattach the device I see that the file was not copied over.

I bought my KUSBA from Amazon from the [Isik's Tech Store](https://www.amazon.com/gp/product/B0C734JL4Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

What can I do to make this work properly?

dvystrcil, 2024-02-03:
I may have gotten this to work. Not sure what changed.

xbst, 2024-02-03:
The file won't show up again, but it'll be in the flash. You should see the serial port when the device isn't in flashing mode.

dvystrcil, 2024-02-03:
Thank you for following up with me. Yes, I understand that part now.

However when I unplug and plug the device back in without pressing the button. It does not show up again. I am forced to reflash and keep it plugged in. Is this expected?

Also is there a specific data cable I need to use? I have a shorter one (6 inches) which works, my longer one (6 feet) does not work with the device. Both of these cables work with my phone.

xbst, 2024-02-03:
Having to reflash after unplugging then pluging the cable back in is a known issue affecting a small percantage of KUSBAs, but KUSBAs I sell on Amazon are tested units without this issue when using Rampon firmware. Klipper is pickier than Rampon so if you're flashing Klipper instead of Rampon it might explain what you're experiencing. Rampon is also less buggy than Klipper, so that's the firmware I recommend, and the firmware I pre-flash. Instructions: https://github.com/xbst/KUSBA/blob/main/Docs/v2-Rampon-Firmware.md (Works with Klipper)

Since all units I sell on Amazon are tested with Rampon to not have this issue, I expect flashing Rampon to solve this issue. But if you still experience similar issues with Rampon, you should be able to return and/or request replacements on Amazon.

All USB C cables with data wires should work.

dvystrcil, 2024-02-03:
That worked. I also popped over to the discord to find clues on how to do x and y separately. I am printing now and have my fingers crossed.

Very shortly I will be updating Klipper, can I keep the existing input shaping or will I need to recalibrate it? I think I can just keep it since nothing about the physical printer is changing.

Thanks again for your help.

xbst, 2024-02-03:
You don't need to recalibrate after Klipper updates.

closed 2024-02-03 by dvystrcil