xbst/KUSBA #32: Downloading adxlmcu.cfg file on Windows 11 messes up formatting

charminULTRA, 2024-02-10:
"Download the [adxlmcu.cfg](https://github.com/xbst/KUSBA/blob/main/Firmware/v2-Rampon/adxlmcu.cfg) file from this repo and add it to your Klipper config directory."

I did this, and used the web interface on Mainsail to upload the file, without any editing, and the text wrapping was completely gone, and it's very hard to read. Is there a way to fix this? If so, I'd recommend a warning in the instructions for PC users.

xbst, 2024-02-10:
I just downloaded this file and it downloaded just fine. Most people download this file on Windows and upload using Mainsail without any problems. Are you sure you didn't download page source or something like that instead of the raw file?

charminULTRA, 2024-02-10:
I right clicked the link and hit Save Link As, which is a typical action
for Windows to save things in their current format. Left clicking the link
brought me to the GitHub page of the file contents which isn’t what I
wanted or expected.

charminULTRA, 2024-02-10:
> I just downloaded this file and it downloaded just fine. Most people download this file on Windows and upload using Mainsail without any problems. Are you sure you didn't download page source or something like that instead of the raw file?

Following up here, when using the Download button on this page linked, it works fine: https://github.com/xbst/KUSBA/blob/main/Firmware/v2-Rampon/adxlmcu.cfg I can't explain why this doesn't work the same as Save Link As.

xbst, 2024-02-11:
It's because "save link as" saves the entire page, not just the file. So it includes everything you see when you click that link (GitHub logo, buttons, etc.) instead of just that file.

closed 2024-02-11 by xbst