xbst/KUSBA #36: Installation problem

shkup, 2024-05-11:
I have Kusba v2.4. I followed the open issues here but I can't solve my issue. It seems the device port can't be connected to.
I already tried to flush it with the Rampon Anchor 0.4 to no avail. 
In the `menuconfig` I selected "Raspberry PI RP2040" and as communication interface: "USBSERIAL".
Here are some outputs:
```
 dmesg | tail
[ 1091.945579] usb 1-1.4: Manufacturer: Anchor
[ 1091.950408] cdc_acm 1-1.4:1.0: ttyACM1: USB ACM device
[ 1212.095035] usb 1-1.5: USB disconnect, device number 6
[ 1212.400028] usb 1-1.5: new full-speed USB device number 13 using dwc_otg
[ 1212.534364] usb 1-1.5: New USB device found, idVendor=1d50, idProduct=614e, bcdDevice= 1.00
[ 1212.534389] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1212.534397] usb 1-1.5: Product: rp2040
[ 1212.534403] usb 1-1.5: Manufacturer: Klipper
[ 1212.534409] usb 1-1.5: SerialNumber: 455035712894B258
[ 1212.535598] cdc_acm 1-1.5:1.0: ttyACM2: USB ACM device
```
For `lsusb` command:
```
 lsusb
Bus 001 Device 013: ID 1d50:614e OpenMoko, Inc. rp2040
Bus 001 Device 012: ID 1d50:614e OpenMoko, Inc. Rampon
Bus 001 Device 009: ID 1d50:614e OpenMoko, Inc. stm32f042x6
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
For `ls /dev/serial/by-id/` command:
```
usb-Anchor_Rampon-if00  usb-Klipper_rp2040_455035712894B258-if00  usb-Klipper_stm32f042x6_120001800743564E32313720-if00
```

From the command: ` sudo mount /dev/sda1 /mnt`
I get: `mount: /mnt: special device /dev/sda1 does not exist.`
I tried switching cables.
I also try to run a python code to attempt to read data from the device:
```
import serial
import time

# Set up serial connection (ensure the baud rate matches your device's configuration)
ser = serial.Serial('/dev/ttyACM2', 115200, timeout=1)

# Read from the device
try:
    while True:
        data = ser.readline()  # Read a line from the serial port
        if data:
            print(data.decode().strip())  # Print the data received from the device
        time.sleep(1)
except KeyboardInterrupt:
    print("Stopped by user")

# Close the serial connection
ser.close()
```
And get the error:
```
 File "/home/shkop/klipper/read_serial.py", line 10, in <module>
    data = ser.readline()  # Read a line from the serial port
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 577, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```
This is the config file:
```
[mcu adxl]
serial: /dev/serial/by-id/usb-Klipper_rp2040_455035712894B258-if00
# Edit the above line with the correct serial address. Use "ls /dev/serial/by-id/*" to find the MCU serial address.

[adxl345]
cs_pin: adxl:gpio1

# Hardware SPI Option (Better results, "Invalid ID" Klipper bug with first query)
spi_bus: spi0a

# Software SPI Option (No "Invalid ID" Klipper bug with first query)
# spi_software_sclk_pin: adxl:gpio2
# spi_software_mosi_pin: adxl:gpio3
# spi_software_miso_pin: adxl:gpio0

[resonance_tester]
accel_chip: adxl345
probe_points:
   60,60,20
# Edit the above line with the correct probe points location. I recommend the center of your bed for X & Y, 20 for Z.
# For 350 mm printers: 175,175,20
# For 300 mm printers: 150,150,20
# For 250 mm printers: 125,125,20
# For 120 mm printers: 60,60,20

# More info: https://www.klipper3d.org/Config_Reference.html#adxl345
```

Any idea what I can do to solve it?

shkup, 2024-05-12:
I think I sorted it out. 
I think the firmware flashing routine is wrong. 
On `klipper` documentation they suggest the following: (All the commands are `bash` ones you can run with `SSH` from `PuttY` except the `ACCELEROMETER_QUERY` that you un from `mainsail` console).

1. Download the `klipper` usb firmware [here](https://github.com/bigtreetech/SKR-Pico/blob/master/Klipper/klipper-USB.uf2)
2. Connect the `ADXL345` by a `usb` cable to the PC while pressing the boot button and copy the `u2f` file from stage 1, to the `ADXL345` device. (It will be identified as a storage device and open it as `RPI-RP2` folder).
3. While holding down the BOOT button on the device, connect it to the Raspberry Pi via USB. 
4. Compile and flash the firmware in the device
```bash
make flash FLASH_DEVICE=first
```
If that fails, you will be told which FLASH_DEVICE to use. In this example, that's make flash FLASH_DEVICE=2e8a:0003.
After that you suppose to see the following output:
```
 Flashing out/klipper.uf2 to 2e8a:0003
sudo lib/rp2040_flash/rp2040_flash out/klipper.uf2

Loaded UF2 image with 109 pages
Found rp2040 device on USB bus 1 address 12
Flashing...
Resetting interface
Locking
Exiting XIP mode
Erasing
Flashing
Rebooting device
```
5. Now you can list the usb devices ids using `ls /dev/serial/by-id/*` command. If you have multiple entries and you are not sure who is the device: run `lsusb` command, disconnect the `ADXL345` usb, and run `lsusb` again. The missing id is your `ADXL345` device.
6. Make sure SPI is enabled in the raspberry pi. run the command: `sudo raspi-config`, Select Interfacing Options > SPI and enable it.
7. Copy the `ADXL345` board id from the `ls /dev/serial/by-id/*` command into your `adxl.cfg` serial id.
8. Restart `klipper` for good measure.
9. You should be able to send now `ACCELEROMETER_QUERY` commands in `mainsail` console.

Good Luck.

xbst, 2024-05-14:
I know the naming can be a bit confusing, but Rampon is fully compatible with Klipper firmware running on the printer's SBC (Pi, etc). Klipper firmware can be flashed to the KUSBA as well, but it is not recommended as it is buggier and more difficult to flash. However, for people who prefer to flash Klipper instead, the prefered method would be to build the firmware yourself because Klipper updates tend to break compatibility of previous versions. Linking a prebuilt firmware by BTT is not ideal because Klipper updates will likely break it soon.

The whole sda1 part can be worded better, I agree. What that part is about, is finding the KUSBA appearing as a USB drive so you can flash the firmware. Using `make flash FLASH_DEVICE=first` will likely result in people flashing the firmware they built for KUSBA to some other device. So, even though its more difficult, finding the correct sdx and flashing to that is preferable.

There is another way. Copying the Klipper.bin file to a computer, and just dragging and dropping it into the KUSBA, which will appear like a USB drive. This is how Rampon is flashed too. Much easier, quicker, less confusing. But the user needs to know how to copy the Klipper.bin file from the Pi to the computer, which a lot of users don't. So it's not a part of the Klipper instructions, only Rampon.

Most KUSBAs these days come with Rampon pre-flashed (looks like yours did as well) so there's no reason to do any of these. Just add the `adxlmcu.cfg` file, and should be ready to use. Enabling SPI on the Raspberry Pi (step 6) is also not necessary. We are not using the Pi's SPI bus (on its GPIO pins).

I understand the instructions are not perfect, and I appreciate the help, but in this case, this will only make the problem worse when a Klipper update breaks compatibility. I will try to come up with a better way of wording the sda1 part, and add further clarificarion about Klipper vs Rampon to avoid confusion.

shkup, 2024-05-14:
I agree but I couldn't flash the device as there was no SDA folder created no matter what I did.
I did try dropping the rampon firmware in the device using the PC method but still no SDA was present.
I don't know if I got a faulty USB device unit. The lack of status LEDs really makes it harder to diagnose.
I will try to flush the rampon firmware again.
Of course  it would be clearer if it would be explained that rampon firmware is compatible with Klipper.
Maybe you can add a troubleshooting section as it seems I'm not the only one struggling with it.

Thank you, Yaron.