xbst/KUSBA #37: KUSBA Unknown spi_bus 'spi0a'

CyberKK, 2024-07-09:
Hi,
I'm trying to set up my KUSBA with Klipper but I get the following error in Mainsail: 

Klipper reports: ERROR

Unknown spi_bus 'spi0a'

I have followed the docs instructions but I can't get it to work. The KUSBA is showing when I SSH in with putty as /dev/serial/by-id/usb-Anchor_Rampon-if00 but it doesn't show up in Mainsail. The issue is with the KUSBA as if I comment out adxlmcu,cfg in the printer.cfg, then no issue occurs. I'm guessing I'm doing something wrong during install.

I've included the printer and adxl config files and the klipper and moonraker logs. I'd appreciate any help you can provide.

[klippy.log](https://github.com/user-attachments/files/16143916/klippy.log)
[moonraker.log](https://github.com/user-attachments/files/16143919/moonraker.log)

[printer.cfg.txt](https://github.com/user-attachments/files/16143882/printer.cfg.txt)
[adxlmcu.cfg.txt](https://github.com/user-attachments/files/16143884/adxlmcu.cfg.txt)

xbst, 2024-07-09:
You need to use this version of adxlmcu.cfg:
https://github.com/xbst/KUSBA/blob/main/Firmware/v2-Rampon/adxlmcu.cfg

CyberKK, 2024-07-09:
Thanks, that got it working.

closed 2024-07-09 by CyberKK